NVIDIA-AI-IOT/Lidar_AI_Solution #237: Does the coordinate system affect the accuracy?

eugenebak, 2024-03-14:
Hi !

I successfully trained the fusion model with KITTI dataset based on bevfusion repo.
But, result of tensorrt engine is totally wrong!!

When I use the LiDAR only for inference, I've noticed that the output of LiDAR-only model (SCN) is NaN.

I've carefully modified the corresponding parameters (point cloud range, voxel size, etc. in `main.cpp` and `python.cpp`).

I wonder it's caused by the different coordinate systems between KITTI and nuScenes...
Or is there any reason which could raise this results..?

Thank you in advance!

eugenebak, 2024-03-16:
@hopef 
Thank you for your response!

But, the coordinate system of KITTI is not just zyx of nuScenes.

If I denote the nuScenes's coordinates system as `xyz`, the KITTI's one is `(-y)xz`.

Can I apply this coordinate system by editing the source code?

eugenebak, 2024-03-16:
I think that I need to modify some parts where the scn model is converted to ONNX.

I'll try this and I'll appreciate it if you say another part I need to modify

closed 2024-05-13 by eugenebak

Li-Whasaka, 2024-05-19:
Hi, Could you share your work with kitti?

eugenebak, 2024-05-28:
@Li-Whasaka Hi.
I'm sorry, that work cannot be shared because of the security terms of the company.
It was quite hard work to fit this repo for KITTI. If you tried it, feel free to any question or something

chfyyds, 2024-06-18:
> @Li-Whasaka Hi. I'm sorry, that work cannot be shared because of the security terms of the company. It was quite hard work to fit this repo for KITTI. If you tried it, feel free to any question or something

Hello, do you need to modify any other files besides the data loading and config? I have successfully run bevfusion on kitti, but the val result is almost 0, I don't know where the problem is, can you give me a useful answer. thank you very much！

chfyyds, 2024-06-18:
@eugenebak 
Hello, do you need to modify any other files besides the data loading and config? I have successfully run bevfusion on kitti, but the val result is almost 0, I don't know where the problem is, can you give me a useful answer. thank you very much！

PW812, 2025-01-02:
> Hi !
> 
> I successfully trained the fusion model with KITTI dataset based on bevfusion repo. But, result of tensorrt engine is totally wrong!!
> 
> When I use the LiDAR only for inference, I've noticed that the output of LiDAR-only model (SCN) is NaN.
> 
> I've carefully modified the corresponding parameters (point cloud range, voxel size, etc. in `main.cpp` and `python.cpp`).
> 
> I wonder if it's caused by the different coordinate systems between KITTI and nuScenes... Or is there any reason which could raise this results..?
> 
> Thank you in advance!

Hello！
Have you solved this problem? I am facing the same issue. Could you share your approach or method for resolving it?
Thank you in advance!

PW812, 2025-01-02:
> @eugenebak Hello, do you need to modify any other files besides the data loading and config? I have successfully run bevfusion on kitti, but the val result is almost 0, I don't know where the problem is, can you give me a useful answer. thank you very much！

Hello！
Have you solved this problem? I am facing the same issue. Could you share your approach or method for resolving it?
Thank you in advance!