NVIDIA-AI-IOT/Lidar_AI_Solution #238: Building the TRT engines via C++ API, how to handle int8 calibration?

josh-wende, 2024-03-14:
I am trying to re-implement the `build_trt_engine.sh` script using the TensorRT C++ API instead of `trtexec`. When I try to build the engines for `resnet50int8`, I get errors due to not setting up an int8 calibrator. But the `trtexec` method seems to build the engine without any calibration data, and doesn't specify anything about int8 calibration. Is there a step I am missing?

hopef, 2024-03-15:
https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/ptq.py#L136

josh-wende, 2024-03-15:
I think the issue in this case was just that the `camera.vtransform.onnx` and `head.bbox.onnx` models have no Quantize/Dequantize layers in them, so TensorRT's C++ API does not automatically recognize them as explicitly quantized and expects an int8 calibrator then. I was able to get the models to build by just not setting the `BuilderFlag::kINT8` flag for those two models.

closed 2024-03-15 by josh-wende