NVIDIA-AI-IOT/Lidar_AI_Solution #239: Half precision error when BEVFusion network built via C++ API instead of `trtexec`

josh-wende, 2024-03-14:
I am trying to re-implement the BEVFusion `build_trt_engine.sh` script using the TensorRT C++ API instead of trtexec. I can build the engines for `resnet50`, but when I try to run them, the assert in `init` in `tranfusion.cu` fails, as the network has FP32 output binding type instead of FP16. The inputs also have FP32. Am I missing a step?

Any insight is greatly appreciated,

hopef, 2024-03-15:
Are you adding the --inputIOFormats/--outputIOFormats flags when building the engine?

https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/d08fc46bfc0bcbae4e965a3cd970c0b7f7f72ec1/CUDA-BEVFusion/tool/build_trt_engine.sh#L84

josh-wende, 2024-03-18:
Yes this is what I was struggling to replicate via the API. I added these lines:

```
    for (int i = 0; i < numInputs; ++i) {
      network_->getInput(i)->setType(nvinfer1::DataType::kHALF);
      network_->getInput(i)->setAllowedFormats(
          1U << static_cast<int>(nvinfer1::TensorFormat::kCHW16));
    }
    for (int i = 0; i < numOutputs; ++i) {
      network_->getOutput(i)->setType(nvinfer1::DataType::kHALF);
      network_->getOutput(i)->setAllowedFormats(
          1U << static_cast<int>(nvinfer1::TensorFormat::kCHW16));
    }
```

and got it to successfully build and run. Still not outputting the correct values, but I think this issue is resolved.

closed 2024-03-18 by josh-wende