NVIDIA-AI-IOT/Lidar_AI_Solution #243: export-scn.py Error:

Bonasdljlkj, 2024-03-28:
(mitbevfusion) gss@Gss:~/Lidar_AI_Solution/CUDA-BEVFusion$ python qat/export-scn.py 
Tracing model inference...
> Do inference...
   --> SparseConvolutionQunat0[subm] -> Input 0, Output 1
Traceback (most recent call last):
  File "qat/export-scn.py", line 73, in <module>
    exptool.export_onnx(model, voxels, coors, batch_size, inverse_indices, args.save)
  File "/home/gss/Lidar_AI_Solution/CUDA-BEVFusion/qat/lean/exptool.py", line 401, in export_onnx
    y = model(voxels, coors, batch_size)
  File "/home/gss/anaconda3/envs/mitbevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/gss/Lidar_AI_Solution/CUDA-BEVFusion/qat/lean/exptool.py", line 275, in impl
    x = self.conv_input(input_sp_tensor)
  File "/home/gss/anaconda3/envs/mitbevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/gss/Lidar_AI_Solution/CUDA-BEVFusion/qat/lean/exptool.py", line 62, in internal_forward
    bind_fn(self, ilayer, y, *args)
  File "/home/gss/Lidar_AI_Solution/CUDA-BEVFusion/qat/lean/exptool.py", line 179, in symbolic_sparse_convolution_quant
    input_dynamic_range  = self._input_quantizer.amax.cpu().item(),
AttributeError: 'NoneType' object has no attribute 'cpu'

Felix-fz, 2024-07-16:
I have the same problem. Did you solve it？

stephankee, 2025-03-26:
you can check the official lidar onnx model, where the input_dynamic_range and input0_dynamic_range do not exist, so just comment out the 'input_dynamic_range = self._input_quantizer.amax.cpu().item(),' and related codes, the script will run successfully

pacientes, 2025-03-26:
> you can check the official lidar onnx model, where the input_dynamic_range and input0_dynamic_range do not exist, so just comment out the 'input_dynamic_range = self._input_quantizer.amax.cpu().item(),' and related codes, the script will run successfully

is it correct solution?
apply after this, start tool/run.sh script get failed to create lidar scn error

stephankee, 2025-03-26:
> is it correct solution? apply after this, start tool/run.sh script get failed to create lidar scn error

for my case, i get correct inference result after bash tool/build_trt_engine.sh and bash tool/run.sh, maybe you can compare the exported onnx model with the official onnx model and see if these models have same structure?

pacientes, 2025-03-26:
> for my case, i get correct inference result after bash tool/build_trt_engine.sh and bash tool/run.sh, maybe you can compare the exported onnx model with the official onnx model and see if these [...]

thanks,

i trained custom datasets for the model where mmdet3d official repo (not bevfusion-mit) model.
so when ptq step, skip the quantize and calibrate code

```
    model = load_model(cfg, checkpoint_path = args.ckpt)
    # model = quantize_net(model)
    # model = fuse_conv_bn(model)
    model = MMDataParallel(model, device_ids=[0])
    model.eval()

    ##Calibrate
    print("🔥 start calibrate 🔥 ")
    # quantize.set_quantizer_fast(model)
    # quantize.calibrate_model(model, data_loader_train, 0, None, args.calibrate_batch)
    
    # quantize.disable_quantization(model.module.encoders.lidar.backbone.conv_input).apply()
    # quantize.disable_quantization(model.module.decoder.neck.deblocks[0][0]).apply()
    # quantize.print_quantizer_status(model)
    
    print(f"Done due to ptq only! Save checkpoint to {save_path} 🤗")
    model.module.encoders.lidar.backbone = funcs.fuse_relu_only(model.module.encoders.lidar.backbone)
    torch.save(model, save_path)
```

so it will fine

pacientes, 2025-03-26:
I dont know why this error occur now.

I remember that using original model and original code it would be fine.
but i tested now, original model occur error too

pacientes, 2025-03-26:
> for my case, i get correct inference result after bash tool/build_trt_engine.sh and bash tool/run.sh, maybe you can compare the exported onnx model with the official onnx model and see if these [...]

have you comment out like this?

```
            output_shape = y.features.shape,
            # input_dynamic_range  = self._input_quantizer.amax.cpu().item(),
            # weight_dynamic_ranges = self._weight_quantizer.amax.cpu().view(-1).numpy().tolist(),
            precision = "fp16" if hasattr(self, "precision") is None else self.precision,
```

stephankee, 2025-03-26:
> have you comment out like this?
>
> ```
>             output_shape = y.features.shape,
>             # input_dynamic_range  = self._input_quantizer.amax.cpu().item(),
>             # weight_dynamic_ranges = [...]

i comment out these four codes:'            
# input_dynamic_range  = self._input_quantizer.amax.cpu().item(),
# weight_dynamic_ranges = self._weight_quantizer.amax.cpu().view(-1).numpy().tolist(),             
# input0_dynamic_range = self._input_quantizer.amax.cpu().item(),
# input1_dynamic_range = self._input_quantizer.amax.cpu().item(),' in file exptool.py, and in your case, the mmdet3d version of bevfusion still need to quantize_net and fuse_conv_bn (as tested before, i only leave out :'# quantize.calibrate_model(model, data_loader_train, 0, None, args.calibrate_batch)').
But the inference result did not look right, model(mmdet3d version) fail to output correct bbox

pacientes, 2025-03-26:
> i comment out these four codes:'
>
> # input_dynamic_range = self._input_quantizer.amax.cpu().item(),
> # weight_dynamic_ranges = self._weight_quantizer.amax.cpu().view(-1).numpy().tolist(),
> # [...]

comment out 4 lines too.

and in my case final ptq.py code is

```
# ptq.py
    model = load_model(cfg, checkpoint_path = args.ckpt)
    model = quantize_net(model)
    model = fuse_conv_bn(model)
    model = MMDataParallel(model, device_ids=[0])
    model.eval()

    ##Calibrate
    print("🔥 start calibrate 🔥 ")
    quantize.set_quantizer_fast(model)
    # quantize.calibrate_model(model, data_loader_train, 0, None, args.calibrate_batch)
    
    quantize.disable_quantization(model.module.encoders.lidar.backbone.conv_input).apply()
    quantize.disable_quantization(model.module.decoder.neck.deblocks[0][0]).apply()
    quantize.print_quantizer_status(model)
    
    print(f"Done due to ptq only! Save checkpoint to {save_path} 🤗")
    model.module.encoders.lidar.backbone = funcs.fuse_relu_only(model.module.encoders.lidar.backbone)
    torch.save(model, save_path)
```

and 
```
# export-scn.py
    for name, module in model.named_modules():
        module.precision = "fp16"
        module.output_precision = "fp16"
    
    model.conv_input.precision = "fp16"
    model.conv_out.output_precision = "fp16"
```

mmdet3d version model result look like occur scale error..

stephankee, 2025-03-26:
do you use the spconv from traveller59? # @pacientes

pacientes, 2025-03-26:
> do you use the spconv from traveller59? # [@pacientes](https://github.com/pacientes)

traveller59 spconv use train step (mmdetection3d builf from source with traveller59 spconv),
tensorrt inference with libspconv from this repo.

stephankee, 2025-03-28:
@pacientes Thanks a lot for sharing! i encountered a problem with the mmdetection3d-bevfusion: the state_dicts' name are different from this repo(mit), for example:'encoders.camera.backbone.patch_embed.projection.weight' in mit version, while 'img_backbone.patch_embed.projection.weight' in mmdetection3d version, did you have this problem?

pacientes, 2025-04-03:
> [@pacientes](https://github.com/pacientes) Thanks a lot for sharing! i encountered a problem with the mmdetection3d-bevfusion: the state_dicts' name are different from this repo(mit), for [...]

yes.
I write first name change function. and it could be resolved.

but mmdet3d version bevfusion transform tensorrt (this repo) version result is poorly bad.
I don't know how emprove this problem.