NVIDIA-AI-IOT/Lidar_AI_Solution #244: Static version of `spconv` library?

josh-wende, 2024-04-01:
I am trying to integrate this solution into an existing repository, which already depends on `libprotobuf.so.23`. This is giving me a lot of trouble, because `libspconv.so` is linked to `libprotobuf.so.17`. As such, whenever these two parts of my code touch, the duplicate symbol names collide and everything crashes. If Nvidia could provide either a) a static version of `libspconv.so` or b) a version of `libspconv.so` linked against newer versions of protobuf, it would make all of this so much easier, and would be much appreciated. Is there any chance of this happening?

hopef, 2024-06-25:
I have updated the new libspconv.so which will be decoupled from libprotobuf.
Please look at it [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution).

windzu, 2024-08-06:
> I have updated the new libspconv.so which will be decoupled from libprotobuf. Please look at it [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution).

I have observed that in the protobuf-decoupled implementation of spconv, some header files have changed compared to the previous version. How should I use the new spconv? Are there any usage references?

windzu, 2024-08-27:
> I have observed that in the protobuf-decoupled implementation of spconv, some header files have changed compared to the previous version. How should I use the new spconv? Are there any usage [...]

@hopef Hello, is there any new progress on this issue?