NVIDIA-AI-IOT/Lidar_AI_Solution #247: CenterPoint model

super-liuyang, 2024-05-12:
Is there a pre-converted ONNX model available for the CenterPoint .pth model

hopef, 2024-06-24:
We used the official pth model in CenterPoint merely.
You can find the URL from [here](https://github.com/tianweiy/CenterPoint/blob/master/configs/nusc/README.md).

mx2013713828, 2024-06-25:
> We used the official pth model in CenterPoint merely. You can find the URL from [here](https://github.com/tianweiy/CenterPoint/blob/master/configs/nusc/README.md).

Hi,thanks your work. I want to train my custom centerpoint model, can I use openpcdet impl ?  and can i export the model that trained in openpcdet codebase?
waiting for your reply~!

hopef, 2024-06-25:
But openpcdet is a pointpillars-based architecture. 
This is contradictory to the sparse conv we support. 
Are you sure you want to train a pointpillars-based model?

hopef, 2024-06-25:
Yeah, this configuration is pointpillars based centerpoint model. It will be supported by TensorRT.

mx2013713828, 2024-06-25:
> Yeah, this configuration is pointpillars based centerpoint model. It will be supported by TensorRT.

well, I dont know how to export pytorch model to TensorRT model, can I just use export-scn.py?

hopef, 2024-06-25:
No, you can't use the export-scn.py.
This [code ](https://github.com/NVIDIA-AI-IOT/CUDA-PointPillars/blob/main/tool/export_onnx.py) may be helpful to you.

mx2013713828, 2024-06-25:
> No, you can't use the export-scn.py. This [code ](https://github.com/NVIDIA-AI-IOT/CUDA-PointPillars/blob/main/tool/export_onnx.py) may be helpful to you.

Thanks, I got it ,when i use pillar based model ,I can use cuda-pointpillars export code,  but I also need to export the center head ,right?