NVIDIA-AI-IOT/Lidar_AI_Solution #252: Failure in using Torch to execute PTQ checkpoint for BEVFusion inference

barrydoooit, 2024-06-12:
Hi there. I'm trying to conduct inference with the PTQ checkpoint using pytorch. I've tried two ways but the second way failed:

1. Build the BEVFusion model and load the original bevfusion checkpoint bevfusion-det.pth, then do everything in the ptq.py, like:
        
```
#import qat.lean.quantize as qt
#import qat.ptq as ptq

model = ptq.load_model(cfg, args.checkpoint) #bevfusion-det.pth
model = ptq.quantize_net(model)
model = ptq.fuse_conv_bn(model)
model = MMDataParallel(model, device_ids=[0])
model.eval()
qt.set_quantizer_fast(model)
qt.calibrate_model(model, data_loader_calib, 0, None, 300)
qt.disable_quantization(model.module.encoders.lidar.backbone.conv_input).apply()
qt.disable_quantization(model.module.decoder.neck.deblocks[0][0]).apply()
```

2. load the PTQ checkpoint (i.e., the provided bevfusion_ptq.pth)
```
model = torch.load(args.checkpoint).module
model = MMDataParallel(model, device_ids=[0])
model.eval()
```

**The problem is that the second way produces zero mAP**. Am I missing some points in using the quantization toolpack? Looking forward to the helps.

hopef, 2024-06-24:
Sorry, I can't see the suspicious issue in your code. It could be due to something else, which would require further troubleshooting.

barrydoooit, 2024-06-25:
Thanks for the reply though :). Maybe I should consider other quantization tools, while this pytorch-quantization from TensorRT is dedicated for onnx exporting.

closed 2024-06-26 by barrydoooit