NVIDIA-AI-IOT/Lidar_AI_Solution #253: How should I export the data of camera_bevfeat of type const nvtype::half*, because I want to visualize the BEV results of Camera. Thank you very much.

KaysenC, 2024-06-14:
Here is the code. I hope to view the camera_bevfeat data through csv file, it seems that the output dimension is {1x80x180x180}[float16].

const nvtype::half* forward_only(const void* camera_images, const nvtype::half* lidar_points,
                                                         int num_points, void* stream, bool do_normalization) {
    int cappoints = static_cast<int>(capacity_points_);
    if (num_points > cappoints) {
      printf("If it exceeds %d points, the default processing will simply crop it out.\n", cappoints);
    }

    num_points = std::min(cappoints, num_points);

    cudaStream_t _stream = static_cast<cudaStream_t>(stream);
    size_t bytes_points = num_points * param_.lidar_scn.voxelization.num_feature * sizeof(nvtype::half);
    checkRuntime(cudaMemcpyAsync(lidar_points_host_, lidar_points, bytes_points, cudaMemcpyHostToHost, _stream));
    checkRuntime(cudaMemcpyAsync(lidar_points_device_, lidar_points_host_, bytes_points, cudaMemcpyHostToDevice, _stream));

    const nvtype::half* lidar_feature = this->lidar_scn_->forward(lidar_points_device_, num_points, stream);
    nvtype::half* normed_images = (nvtype::half*)camera_images;
    if (do_normalization) {
      normed_images = (nvtype::half*)this->normalizer_->forward((const unsigned char**)(camera_images), stream);
    }
    const nvtype::half* depth = this->camera_depth_->forward(lidar_points_device_, num_points, 5, stream);

    this->camera_backbone_->forward(normed_images, depth, stream);
    const nvtype::half* camera_bev = this->camera_bevpool_->forward(
        this->camera_backbone_->feature(), this->camera_backbone_->depth(), this->camera_geometry_->indices(),
        this->camera_geometry_->intervals(), this->camera_geometry_->num_intervals(), stream);

    const nvtype::half* camera_bevfeat = camera_vtransform_->forward(camera_bev, stream);
    const nvtype::half* fusion_feature = this->transfusion_->forward(camera_bevfeat, lidar_feature, stream);
    return camera_bevfeat;
  }

hopef, 2024-06-24:
You can use the save function in class::Tensor.
For example([link](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/d4f4ba54c7a0313b991284b1c22b1b9dfa66e29f/CUDA-BEVFusion/src/common/tensor.hpp#L136)):
```c++
nv::Tensor::from_data_reference(camera_bevfeat, {1, 2, 3 /* the shape of camera_bevfeat */}, nv::Tensor::DataType::Float16, true).save("camera_bevfeat.tensor");
```

And then, you can load it on Python([link](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/d4f4ba54c7a0313b991284b1c22b1b9dfa66e29f/CUDA-BEVFusion/tool/tensor.py#L52)).
```python
import tensor
camera_bevfeat = tensor.load("camera_bevfeat.tensor")
```

KaysenC, 2024-06-25:
thank you for your reply !