NVIDIA-AI-IOT/Lidar_AI_Solution #255: which model should I use?

mx2013713828, 2024-06-25:
Thanks for your great work, but I really need some advices!
I want to deploy a lidar 3d detector,which model should I choose?
BEVFusion or CenterPoint?
Now, I only get custom lidar dataset ,which model perfom better ?
Waitting for your reply !

hopef, 2024-06-25:
Would you like to deploy a **lidar only** detector?

mx2013713828, 2024-06-25:
Actually, I'm not sure how to choose. I have just started working with 3D object detection, and my experimental setup only includes one LiDAR and three cameras,which have not been calibrated.. I've looked into BEVFusion, but I noticed that it requires cameras from six directions. Therefore, I'm considering whether lidar only detector might be more suitable for me.

hopef, 2024-06-25:
6 cameras are not necessary. You can still change to 3 cameras easily.

hopef, 2024-06-26:
Yeah, you are right.