NVIDIA-AI-IOT/Lidar_AI_Solution #256: Spconv Engine Implementation

barrydoooit, 2024-06-26:
Hi @hopef !

I found a huge gap of inference latency (100ms compared to 30ms) on orin of BEVFusion's lidar-scn backbone, between using **the engine returned by spconv::load_engine_from_onnx** and using **the original pytorch implementation**. Therefore I want to check how is that spconv engine handling the forward process and I wonder where may we get the implementation of that Spconv Engine class (defined in [engine.hpp](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/libraries/3DSparseConvolution/libspconv_cuda11/include/spconv/engine.hpp)? 

I suppose it's in the libspconv, but I got no clue of existence of this class from traveller59's spconv repo. Is it implemented additionally along with this Lidar_AI_Solution project and would not be released? Looking forward to your reply.

hopef, 2024-06-27:
I'm sorry, but unfortunately we have no plan to public the libspconv.so.
The libspconv.so in our repository is independent of other implementations. Therefore, you may not find the class in the third repository.
Please feel free to commit any questions you encounter.

BR,
Thanks

barrydoooit, 2024-06-27:
Hi @hopef,
Thank you for the clarification. It's a pity missing the chance to inspect the techniques used to achieve such significant acceleration. Nevertheless, may I learn about what could be the rationale for your custom spconv inference engine to run 3 times faster than the pytorch implementation? In other words, is the acceleration primarily credited to the refined memory management like in TensorRT, or to your algorithmic improvements in the SparseConv kernels?

hopef, 2024-06-27:
First of all, I want to clarify that there is no leading or different acceleration technology here.
The following might be worth considering:
1. Inference on FP32, FP16 or INT8?  Their speed gap is very significant.
2. Reuse the rule book in different layers is important for inference latency.
3. Some memory reuse strategies will be more friendly.
4. Layer fusion has also been utilized.

barrydoooit, 2024-07-08:
@hopef Thank you so much for the hints (while sorry for the late response). I managed to reuse the rulebook for every 4 consecutive sub-manifold conv layers and indeed achieved a 20% latency reeduction (now at 80-90ms per frame) compared to the previous test with layer fusion & under FP16. 

The latency, though, is still too high. I located three main modules that consume a lot of time, where I suppose your Cpp inference engine might be good at:

1. Output SparseConvTensor creation. The call of replace_feature() function after all the calculation takes nearly the same time as gemm. This can take about 30ms.
2. Necessary Rulebook calculation. There are still 2+3+4 (I/O layer + 3 Spconv Layers + 4 First SubM layer) Rulebooks to calculated, which again takes about 30ms in total. The main overhead here seems the call of SpconvOps.sort_1d_by_key_allocator().
3. The shallow copy of the SparseConvTensor at the beginning of each layer could, totally, take 9ms.

I would like to confirm that if your Spconv engine has made any custom improvements in the above modules. If not, I think the latency could drop with substituting the above parts with cpp implementations (through pybind) from traveller59's libspconv.so.

riteshkhrn, 2024-09-25:
Hello @barrydoooit @hopef I have created an opensource version of the 3DSparseConvolution Library.

https://github.com/riteshkhrn/Lidar_AI_Solution/tree/main/libraries/New3DSparseConvolution

check it out! :)