NVIDIA-AI-IOT/Lidar_AI_Solution #261: Failed to build model model/resnet50int8/build/head.bbox.layernormplugin.plan.

Happynewyear000, 2024-07-05:
[image];

 [E] Uncaught exception detected: Unable to open library: libcustom_layernorm.so due to libcustom_layernorm.so: cannot open shared object file: No such file or directory

wanghuohuo0716, 2024-07-06:
same！ following is the log！ 

A CustomLayerNormalization error?

```
[07/06/2024-12:26:12] [V] [TRT] Parsing node: MatMul_348 [MatMul]
[07/06/2024-12:26:12] [V] [TRT] Searching for input: 880
[07/06/2024-12:26:12] [V] [TRT] Searching for input: 1132
[07/06/2024-12:26:12] [V] [TRT] MatMul_348 [MatMul] inputs: [880 -> (200, 1, 128)[FLOAT]], [1132 -> (128, 128)[FLOAT]], 
[07/06/2024-12:26:12] [V] [TRT] Registering layer: 1132 for ONNX node: 1132
[07/06/2024-12:26:12] [V] [TRT] Registering layer: MatMul_348 for ONNX node: MatMul_348
[07/06/2024-12:26:12] [I] [TRT] MatMul_348: broadcasting input1 to make tensors conform, dims(input0)=[200,1,128][NONE] dims(input1)=[1,128,128][NONE].
[07/06/2024-12:26:12] [V] [TRT] Registering tensor: 882 for ONNX tensor: 882
[07/06/2024-12:26:12] [V] [TRT] MatMul_348 [MatMul] outputs: [882 -> (200, 1, 128)[FLOAT]], 
[07/06/2024-12:26:12] [V] [TRT] Parsing node: Add_349 [Add]
[07/06/2024-12:26:12] [V] [TRT] Searching for input: parent.heads.object.decoder.0.self_attn.out_proj.bias
[07/06/2024-12:26:12] [V] [TRT] Searching for input: 882
[07/06/2024-12:26:12] [V] [TRT] Add_349 [Add] inputs: [parent.heads.object.decoder.0.self_attn.out_proj.bias -> (128)[FLOAT]], [882 -> (200, 1, 128)[FLOAT]], 
[07/06/2024-12:26:12] [V] [TRT] Registering layer: parent.heads.object.decoder.0.self_attn.out_proj.bias for ONNX node: parent.heads.object.decoder.0.self_attn.out_proj.bias
[07/06/2024-12:26:12] [V] [TRT] Registering layer: Add_349 for ONNX node: Add_349
[07/06/2024-12:26:12] [V] [TRT] Registering tensor: 883 for ONNX tensor: 883
[07/06/2024-12:26:12] [V] [TRT] Add_349 [Add] outputs: [883 -> (200, 1, 128)[FLOAT]], 
[07/06/2024-12:26:12] [V] [TRT] Parsing node: Add_350 [Add]
[07/06/2024-12:26:12] [V] [TRT] Searching for input: 796
[07/06/2024-12:26:12] [V] [TRT] Searching for input: 883
[07/06/2024-12:26:12] [V] [TRT] Add_350 [Add] inputs: [796 -> (200, 1, 128)[FLOAT]], [883 -> (200, 1, 128)[FLOAT]], 
[07/06/2024-12:26:12] [V] [TRT] Registering layer: Add_350 for ONNX node: Add_350
[07/06/2024-12:26:12] [V] [TRT] Registering tensor: 884 for ONNX tensor: 884
[07/06/2024-12:26:12] [V] [TRT] Add_350 [Add] outputs: [884 -> (200, 1, 128)[FLOAT]], 
[07/06/2024-12:26:12] [V] [TRT] Parsing node: layernorm_0 [CustomLayerNormalization]
[07/06/2024-12:26:12] [V] [TRT] Searching for input: 884
[07/06/2024-12:26:12] [V] [TRT] Searching for input: parent.heads.object.decoder.0.norm1.weight
[07/06/2024-12:26:12] [V] [TRT] Searching for input: parent.heads.object.decoder.0.norm1.bias
[07/06/2024-12:26:12] [V] [TRT] layernorm_0 [CustomLayerNormalization] inputs: [884 -> (200, 1, 128)[FLOAT]], [parent.heads.object.decoder.0.norm1.weight -> (128)[FLOAT]], [parent.heads.object.decoder.0.norm1.bias -> (128)[FLOAT]], 
[07/06/2024-12:26:12] [I] [TRT] No importer registered for op: CustomLayerNormalization. Attempting to import as plugin.
[07/06/2024-12:26:12] [I] [TRT] Searching for plugin: CustomLayerNormalization, plugin_version: 1, plugin_namespace: 
[07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:726: While parsing node number 351 [CustomLayerNormalization -> "895"]:
[07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:727: --- Begin node ---
[07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:728: input: "884"
input: "parent.heads.object.decoder.0.norm1.weight"
input: "parent.heads.object.decoder.0.norm1.bias"
output: "895"
name: "layernorm_0"
op_type: "CustomLayerNormalization"
attribute {
  name: "axis"
  i: -1
  type: INT
}
attribute {
  name: "epsilon"
  f: 1e-05
  type: FLOAT
}

[07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:729: --- End node ---
[07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:731: ERROR: builtin_op_importers.cpp:5427 In function importFallbackPluginImporter:
[8] Assertion failed: creator && "Plugin not found, are the plugin name, version, and namespace correct?"
[07/06/2024-12:26:12] [E] Failed to parse onnx file
[07/06/2024-12:26:12] [I] Finish parsing network model
[07/06/2024-12:26:12] [E] Parsing model failed
[07/06/2024-12:26:12] [E] Failed to create engine from model or file.
[07/06/2024-12:26:12] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8502] # trtexec --onnx=model/resnet50int8/head.bbox.layernormplugin.onnx --fp16 --inputIOFormats=fp16:chw, --outputIOFormats=fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw, --plugins=libcustom_layernorm.so --saveEngine=model/resnet50int8/build/head.bbox.layernormplugin.plan --memPoolSize=workspace:2048 --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed --exportLayerInfo=model/resnet50int8/build/head.bbox.layernormplugin.json
```

**ChatGPT:**

从日志中可以看出，TensorRT在解析ONNX文件时，未能找到名为 `CustomLayerNormalization` 的插件。这导致了模型构建失败。以下是一些可能的解决方法：

### 1. 检查并编译自定义插件

确保自定义插件 `libcustom_layernorm.so` 已正确编译并在正确的路径中。你可以尝试重新编译插件并确保其路径正确。

```sh
# 假设你有Makefile或CMakeLists.txt文件
make
# 或者使用cmake
cmake .
make
```

确保生成的 `libcustom_layernorm.so` 文件在你的库路径中，并且可以被TensorRT找到。

### 2. 设置插件路径

你可以通过设置 `LD_LIBRARY_PATH` 确保插件可以被找到：

```sh
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/your/plugin_directory
```

wanghuohuo0716, 2024-07-06:
git checkout <last commit>!  can fix the bug!

the orror occurs from the newest commit !!!!

eugenebak, 2024-07-08:
When is the libcustom_layernorm.so created?

Ankitzanzmera, 2024-07-09:
@wanghuohuo0716  Please check computation capability of you GPU by using following command 

`nvidia-smi --query-gpu=compute_cap --format=csv`

it will return something like following
> nvidia-smi --query-gpu=compute_cap --format=csv

and if it is less than 8.0 you will not able to run this model.

slayerlpj, 2024-08-02:
> @wanghuohuo0716 Please check computation capability of your GPU by using following command
> 
> `nvidia-smi --query-gpu=compute_cap --format=csv`
> 
> it will return something like following
> 
> > compute_cap
> > 7.5
> 
> and if it is less than 8.0 you will not able to run this model.

My compute_cap is 8.9. But I have the same problem, How to fix it?

wanghuohuo0716, 2024-08-06:
> @wanghuohuo0716 Please check computation capability of your GPU by using following command
> 
> `nvidia-smi --query-gpu=compute_cap --format=csv`
> 
> it will return something like following
> 
> > compute_cap
> > 7.5
> 
> and if it is less than 8.0 you will not able to run this model.

 I use the agx orin 64G, compute_cap 8.7 is greater than 8.0

Ankitzanzmera, 2024-08-06:
@wanghuohuo0716 Which cuda version are you using ??

wanghuohuo0716, 2024-08-06:
@Ankitzanzmera [image]

evil-master, 2024-08-30:
> same！ following is the log！
> 
> A CustomLayerNormalization error?
> 
> ```
> [07/06/2024-12:26:12] [V] [TRT] Parsing node: MatMul_348 [MatMul]
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: 880
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: 1132
> [07/06/2024-12:26:12] [V] [TRT] MatMul_348 [MatMul] inputs: [880 -> (200, 1, 128)[FLOAT]], [1132 -> (128, 128)[FLOAT]], 
> [07/06/2024-12:26:12] [V] [TRT] Registering layer: 1132 for ONNX node: 1132
> [07/06/2024-12:26:12] [V] [TRT] Registering layer: MatMul_348 for ONNX node: MatMul_348
> [07/06/2024-12:26:12] [I] [TRT] MatMul_348: broadcasting input1 to make tensors conform, dims(input0)=[200,1,128][NONE] dims(input1)=[1,128,128][NONE].
> [07/06/2024-12:26:12] [V] [TRT] Registering tensor: 882 for ONNX tensor: 882
> [07/06/2024-12:26:12] [V] [TRT] MatMul_348 [MatMul] outputs: [882 -> (200, 1, 128)[FLOAT]], 
> [07/06/2024-12:26:12] [V] [TRT] Parsing node: Add_349 [Add]
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: parent.heads.object.decoder.0.self_attn.out_proj.bias
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: 882
> [07/06/2024-12:26:12] [V] [TRT] Add_349 [Add] inputs: [parent.heads.object.decoder.0.self_attn.out_proj.bias -> (128)[FLOAT]], [882 -> (200, 1, 128)[FLOAT]], 
> [07/06/2024-12:26:12] [V] [TRT] Registering layer: parent.heads.object.decoder.0.self_attn.out_proj.bias for ONNX node: parent.heads.object.decoder.0.self_attn.out_proj.bias
> [07/06/2024-12:26:12] [V] [TRT] Registering layer: Add_349 for ONNX node: Add_349
> [07/06/2024-12:26:12] [V] [TRT] Registering tensor: 883 for ONNX tensor: 883
> [07/06/2024-12:26:12] [V] [TRT] Add_349 [Add] outputs: [883 -> (200, 1, 128)[FLOAT]], 
> [07/06/2024-12:26:12] [V] [TRT] Parsing node: Add_350 [Add]
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: 796
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: 883
> [07/06/2024-12:26:12] [V] [TRT] Add_350 [Add] inputs: [796 -> (200, 1, 128)[FLOAT]], [883 -> (200, 1, 128)[FLOAT]], 
> [07/06/2024-12:26:12] [V] [TRT] Registering layer: Add_350 for ONNX node: Add_350
> [07/06/2024-12:26:12] [V] [TRT] Registering tensor: 884 for ONNX tensor: 884
> [07/06/2024-12:26:12] [V] [TRT] Add_350 [Add] outputs: [884 -> (200, 1, 128)[FLOAT]], 
> [07/06/2024-12:26:12] [V] [TRT] Parsing node: layernorm_0 [CustomLayerNormalization]
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: 884
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: parent.heads.object.decoder.0.norm1.weight
> [07/06/2024-12:26:12] [V] [TRT] Searching for input: parent.heads.object.decoder.0.norm1.bias
> [07/06/2024-12:26:12] [V] [TRT] layernorm_0 [CustomLayerNormalization] inputs: [884 -> (200, 1, 128)[FLOAT]], [parent.heads.object.decoder.0.norm1.weight -> (128)[FLOAT]], [parent.heads.object.decoder.0.norm1.bias -> (128)[FLOAT]], 
> [07/06/2024-12:26:12] [I] [TRT] No importer registered for op: CustomLayerNormalization. Attempting to import as plugin.
> [07/06/2024-12:26:12] [I] [TRT] Searching for plugin: CustomLayerNormalization, plugin_version: 1, plugin_namespace: 
> [07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:726: While parsing node number 351 [CustomLayerNormalization -> "895"]:
> [07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:727: --- Begin node ---
> [07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:728: input: "884"
> input: "parent.heads.object.decoder.0.norm1.weight"
> input: "parent.heads.object.decoder.0.norm1.bias"
> output: "895"
> name: "layernorm_0"
> op_type: "CustomLayerNormalization"
> attribute {
>   name: "axis"
>   i: -1
>   type: INT
> }
> attribute {
>   name: "epsilon"
>   f: 1e-05
>   type: FLOAT
> }
> 
> [07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:729: --- End node ---
> [07/06/2024-12:26:12] [E] [TRT] ModelImporter.cpp:731: ERROR: builtin_op_importers.cpp:5427 In function importFallbackPluginImporter:
> [8] Assertion failed: creator && "Plugin not found, are the plugin name, version, and namespace correct?"
> [07/06/2024-12:26:12] [E] Failed to parse onnx file
> [07/06/2024-12:26:12] [I] Finish parsing network model
> [07/06/2024-12:26:12] [E] Parsing model failed
> [07/06/2024-12:26:12] [E] Failed to create engine from model or file.
> [07/06/2024-12:26:12] [E] Engine set up failed
> &&&& FAILED TensorRT.trtexec [TensorRT v8502] # trtexec --onnx=model/resnet50int8/head.bbox.layernormplugin.onnx --fp16 --inputIOFormats=fp16:chw, --outputIOFormats=fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw, --plugins=libcustom_layernorm.so --saveEngine=model/resnet50int8/build/head.bbox.layernormplugin.plan --memPoolSize=workspace:2048 --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed --exportLayerInfo=model/resnet50int8/build/head.bbox.layernormplugin.json
> ```
> 
> **ChatGPT:**
> 
> 从日志中可以看出，TensorRT在解析ONNX文件时，未能找到名为 `CustomLayerNormalization` 的插件。这导致了模型构建失败。以下是一些可能的解决方法：
> ### 1. 检查并编译自定义插件
> 
> 确保自定义插件 `libcustom_layernorm.so` 已正确编译并在正确的路径中。你可以尝试重新编译插件并确保其路径正确。
> 
> ```shell
> # 假设你有Makefile或CMakeLists.txt文件
> make
> # 或者使用cmake
> cmake .
> make
> ```
> 
> 确保生成的 `libcustom_layernorm.so` 文件在你的库路径中，并且可以被TensorRT找到。
> ### 2. 设置插件路径
> 
> 你可以通过设置 `LD_LIBRARY_PATH` 确保插件可以被找到：
> 
> ```shell
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/your/plugin_directory
> ```

我也遇到相同的问题，请问你是怎么解决的

WelY1, 2025-02-24:
> git checkout last commit ! can fix the bug!
> 
> the orror occurs from the newest commit !!!!

Which branch does the checkout go to？Please

WangYichen6120, 2025-06-22:
请问该libcustom_layernorm.so应放置在什么路径上呢