NVIDIA-AI-IOT/Lidar_AI_Solution #265: Support for H100 and sm_90

maueki, 2024-07-18:
Hi,

Does this project work on the H100? 
It seems that libspconv.so does not support sm_90. Is it possible to get a binary that supports it?

```
$ cuobjdump --list-elf libspconv.so
ELF file    1: libspconv.1.sm_80.cubin
ELF file    2: libspconv.2.sm_86.cubin
ELF file    3: libspconv.3.sm_80.cubin
ELF file    4: libspconv.4.sm_86.cubin
ELF file    5: libspconv.5.sm_80.cubin
ELF file    6: libspconv.6.sm_86.cubin
ELF file    7: libspconv.7.sm_80.cubin
ELF file    8: libspconv.8.sm_86.cubin
ELF file    9: libspconv.9.sm_80.cubin
ELF file   10: libspconv.10.sm_86.cubin
```

Thank you.

Icecream-blue-sky, 2024-09-11:
Same problem. Current libspconv.so doesn't support H100 sm_90：
-- Configuring done (0.1s)
-- Generating done (0.1s)
-- Build files have been written to: /perception/icecream/code/Lidar_AI_Solution-master/CUDA-BEVFusion/build
[  3%] Linking CXX shared library libcustom_layernorm.so
[  7%] Linking CXX shared library libbevfusion_core.so
[ 11%] Built target custom_layernorm
[ 70%] Built target bevfusion_core
[ 74%] Linking CXX executable bevfusion
[100%] Built target bevfusion
Create by resnet50int8, int8
CUDA Runtime error init_quantization_scale<FeaturesType::INT8><<<grid, block, 0, _stream>>>( input_scale, output_scale, weight_, prod_DHWI, int8weight_.ptr(), weight_scales_.ptr(), alpha_.ptr(), (half *)bias, bias_scaled_.ptr(), numel) # no kernel image is available for execution on the device, code = cudaErrorNoKernelImageForDevice [ 209 ] in file src/spconv/implicit-gemm.cu:241
tool/run.sh: line 44: 2057630 Aborted                 (core dumped) ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION

hopef, 2024-09-11:
Hi,

libspconv with SM90 has been supported.

```bash
$> cuobjdump --list-elf Lidar_AI_Solution/libraries/3DSparseConvolution/libspconv/lib/x86_64_cuda12.6/libspconv.so
ELF file    1: libspconv.1.sm_80.cubin
ELF file    2: libspconv.2.sm_86.cubin
ELF file    3: libspconv.3.sm_90.cubin
ELF file    4: libspconv.4.sm_80.cubin
ELF file    5: libspconv.5.sm_86.cubin
ELF file    6: libspconv.6.sm_90.cubin
ELF file    7: libspconv.7.sm_80.cubin
ELF file    8: libspconv.8.sm_86.cubin
ELF file    9: libspconv.9.sm_90.cubin
ELF file   10: libspconv.10.sm_80.cubin
ELF file   11: libspconv.11.sm_86.cubin
ELF file   12: libspconv.12.sm_90.cubin
ELF file   13: libspconv.13.sm_80.cubin
ELF file   14: libspconv.14.sm_86.cubin
ELF file   15: libspconv.15.sm_90.cubin
```