NVIDIA-AI-IOT/Lidar_AI_Solution #267: an illegal memory access was encountered

erzhu222, 2024-07-23:
when I try to inference bevfusion model on more than 800000 points, an error occurred and core dumped: 

CUDA Runtime error cudaStreamSynchronize(stream) # an illegal memory access was encountered, code = cudaErrorIllegalAddress [ 700 ] in file src/spconv/engine.cu:787
Aborted (core dumped)

could you please help me with the problem, thanks a lot!

erzhu222, 2024-07-23:
Warning: Found the number of input[216288] points exceeds the theoretical maximum number[200000], the default processing is to truncate.

hopef, 2024-07-24:
Hi, could you share the model and data with me to reproduce this issue?

erzhu222, 2024-07-24:
> Hi, could you provide me with the model and data so that I can reproduce this issue?

thanks for your reply，the core dump problem was solved by setting max_voxels bigger，you can use more than 1000000 points to reproduce the warning, and I have a question about theoretical maximum number[200000], what does that mean? It seems in sparse conv lib

wangziyi2016, 2025-06-11:
@hopef 
Hi, we’re facing the same issue:
`Warning: Found the number of input[150288] points exceeds the theoretical maximum number[100000], the default processing is to truncate.`
Could you kindly point out where to change the maximum number of input points? We’d like to avoid truncation since point clouds are usually larger than 100,000 points.

Thanks in advance!