NVIDIA-AI-IOT/Lidar_AI_Solution #275: How to  export the SwinT model from BEVFusion?

slayerlpj, 2024-08-02:
I have trained camera+lidar based on SwinT, I have got a pth. How can I generate PTQ model or FP16 model?

I used the command "python qat/ptq.py --config=configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/configs.yaml --ckpt=model/swint/epoch_30.pth --calibrate_batch 300", the I got a error as follow:

2024-08-02 09:16:14,038 - mmdet - INFO - load checkpoint from http path: https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth
I0802 09:16:14.038485 139857851803456 logging.py:101] load checkpoint from http path: https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth
load checkpoint from local path: model/swint/epoch_30.pth
Traceback (most recent call last):
  File "qat/ptq.py", line 150, in <module>
    main()
  File "qat/ptq.py", line 130, in main
    model = quantize_net(model)
  File "qat/ptq.py", line 81, in quantize_net
    quantize.quantize_encoders_camera_branch(model.encoders.camera)
  File "/home/work/work/lpj/BEVFusion/Lidar_AI_Solution/CUDA-BEVFusion/qat/lean/quantize.py", line 175, in quantize_encoders_camera_branch
    major = model_camera_branch.backbone.layer3[0].conv1._input_quantizer
  File "/home/work/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1177, in __getattr__
    raise AttributeError("'{}' object has no attribute '{}'".format(
AttributeError: 'SwinTransformer' object has no attribute 'layer3'

How can I fix the problem?

slayerlpj, 2024-08-02:
When I used the command to export fp16 model, I got  a another error as follow:

python qat/export-camera.py --ckpt=model/swint/epoch_30.pth --fp16
Traceback (most recent call last):
  File "qat/export-camera.py", line 138, in <module>
    main()
  File "qat/export-camera.py", line 85, in main
    model  = torch.load(args.ckpt).module
AttributeError: 'dict' object has no attribute 'module'

evil-master, 2024-09-01:
> 当我使用命令导出 fp16 模型时，我收到了另一个错误，如下所示：
> 
> python qat/export-camera.py --ckpt=model/swint/epoch_30.pth --fp16 回溯（最近调用最后一次）：文件 “qat/export-camera.py”，第 138 行，在 main（） 中 文件“qat/export-camera.py”，第 85 行，在 main model 中 = torch.load（args.ckpt）.module AttributeError：“dict”对象没有属性“module”

how do you reslove it？