NVIDIA-AI-IOT/Lidar_AI_Solution #278: export-transfuser erro

evil-master, 2024-08-30:
我基于resnet50，训练了自己的bevfusion模型，在export onnx过程中遇到了无法转出boxheader部分，报错如下：
我的环境是
onnx==1.12.0
onnxruntim==1.13.1

[image]

hopef, 2024-09-11:
Please ignore those warnings, thanks.

betterhalfwzm, 2025-03-06:
> 我基于resnet50，训练了自己的bevfusion模型，在export onnx过程中遇到了无法转出boxheader部分，报错如下： 我的环境是 onnx==1.12.0 onnxruntim==1.13.1
> 
> [image]

你好，遇到同样问题，解决了吗

evil-master, 2025-03-06:
作者说了忽略就行，warning不碍事

大力金刚
***@***.***

    

 
image.png (view on web)
  

  
 
image.png (view on web)

betterhalfwzm, 2025-03-06:
> 作者说了忽略就行，warning不碍事
> 
> 
> 
> 
> 大力金刚
> ***@***.***
> 
> 
> 
> &nbsp;
> […](#)

但是我忽略了转trt转不过去，加个微信13032215182.谢谢

evil-master, 2025-03-06:
太久了我忘记细节了，当时记录的笔记是量化用nus数据集和config，权重用自己的就行，你试试

大力金刚
***@***.***

betterhalfwzm, 2025-03-06:
> 太久了我忘记细节了，当时记录的笔记是量化用nus数据集和config，权重用自己的就行，你试试
> 
> 
> 
> 
> 大力金刚
> ***@***.***
> 
> 
> 
> &nbsp;
> […](#)

好的，谢谢了