NVIDIA-AI-IOT/Lidar_AI_Solution #279: 预测我自己训练的模型

evil-master, 2024-09-01:
你好，我基于自己的数据集训练了以resnet50的bevfusion，并可视化看了识别效果是正确的，但是转为onnx后识别结果差别非常大，我对比了数据的输入的参数是一致的，对比了输出结果，box的size差距非常大
[image]

此外有点差别的是我在将pth转成PTQ文件时，有报出一些错误：
[image]

并且camera部分需要改成opset=11才能正常输出onnx文件：
[image]

请问一下，差距是由于模型转换过程中造成的吗，我该怎么去处理它呢？下面是我的识别结果，框s看上去太奇怪了：
[image]

evil-master, 2024-09-03:
> 你好，我基于自己的数据集训练了以resnet50的bevfusion，并可视化看了识别效果是正确的，但是转为onnx后识别结果差别非常大，我对比了数据的输入的参数是一致的，对比了输出结果，box的size差距非常大 [...]

我这边已经发现问题的原因；问题出现在量化过程，我这边使用nuscenes数据集和配置文件对我的权重PTQ量化后识别正常

closed 2024-09-19 by mchi-zg

Anmidy, 2024-12-16:
> 你好，我基于自己的数据集训练了以resnet50的bevfusion，并可视化看了识别效果是正确的，但是转为onnx后识别结果差别非常大，我对比了数据的输入的参数是一致的，对比了输出结果，box的size差距非常大 [...]

你好，请问训练resnet50为backbone用的工程是mit版本的代码还是用cuda-bevfusion中的工程代码？配置文件用的那个？我用mit版本的代码和cuda-bevfusion中的configs，训练出来的bevfusion-det 用 ptq.py 量化是报错，KeyError: "NuScenesDataset: 'LoadRadarPointsMultiSweeps is not in the pipeline registry'"

ZFcvYes, 2025-02-07:
> 你好，请问训练resnet50为backbone用的工程是mit版本的代码还是用cuda-bevfusion中的工程代码？配置文件用的那个？我用mit版本的代码和cuda-bevfusion中的configs，训练出来的bevfusion-det 用 ptq.py 量化是报错，KeyError: "NuScenesDataset: 'LoadRadarPointsMultiSweeps is [...]

请问您head.bbox.onnx转成engine可以成功吗？我使用官方提供的model/resnet50int8/bevfusion_ptq.pth权重，使用export-transfuser.py转成onnx后，与官方提供的不一致，且无法使用build_trt_engine.sh转成功。

ZFcvYes, 2025-02-07:
> 我这边已经发现问题的原因；问题出现在量化过程，我这边使用nuscenes数据集和配置文件对我的权重PTQ量化后识别正常

请问您head.bbox.onnx转成engine可以成功吗？我使用官方提供的model/resnet50int8/bevfusion_ptq.pth权重，使用export-transfuser.py转成onnx后，与官方提供的不一致，且无法使用build_trt_engine.sh转成功。