NVIDIA-AI-IOT/Lidar_AI_Solution #281: libspconv source code

jongsik-moon, 2024-09-05:
Would you please open source the spconv code? It would be very helpful for the community.

riteshkhrn, 2024-09-25:
Hello @jongsik-moon I have created an opensource version of the 3DSparseConvolution Library.

https://github.com/riteshkhrn/Lidar_AI_Solution/tree/main/libraries/New3DSparseConvolution

check it out! :)

lysukhin, 2024-11-11:
@riteshkhrn thanks a lot for sharing the code! 
Can I ask you to make the source code of the "libscn.so" library publicly available as well?
The provided .so was (apparently) compiled using CUDA 12 which makes it incompatible with earlier versions of CUDA.
(Sorry for posting in here -- your project page doesn't offer to add issues as far as I can see)

riteshkhrn, 2024-11-11:
@lysukhin The source code and library both are public. you can change the CUDA version in the [Makefile](https://github.com/riteshkhrn/Lidar_AI_Solution/blob/main/libraries/New3DSparseConvolution/Makefile)

_(Sorry for posting in here -- your project page doesn't offer to add issues as far as I can see)_
I have enabled the issues page

lysukhin, 2024-11-12:
@riteshkhrn thanks a lot for the quick reply!

Tony-syr, 2025-03-11:
> Hello [@jongsik-moon](https://github.com/jongsik-moon) I have created an opensource version of the 3DSparseConvolution Library.
> 
> https://github.com/riteshkhrn/Lidar_AI_Solution/tree/main/libraries/New3DSparseConvolution
> 
> check it out! :)

Hello @riteshkhrn , thanks for your work
Could you please explain how to use your opensource version to create libspconv.so? should we install spconv and cumm?
I tried to run make directly but it gives some error like missing headers.