NVIDIA-AI-IOT/Lidar_AI_Solution #286: add  node  cause  result wrong!

NotOnlyAI, 2024-09-27:
identity = x.features

        out = self.conv1(x)
        out = out.replace_feature(self.relu1(out.features))

        out = self.conv2(out)
        out = out.replace_feature(out.features+identity)
        out = out.replace_feature(self.relu2(out.features))

if without  "out = out.replace_feature(out.features+identity)  "  ,the c++ and pytroch result is same .
if has out = out.replace_feature(out.features+identity)   result is not same ?
why?

NotOnlyAI, 2024-09-27:
else if (node.op_type() == "Add" || node.op_type() == "QuantAdd") {
            auto a = tensor_map_by_name[node.input(0)];
            auto b = tensor_map_by_name[node.input(1)];

            auto n = builder->push_add(
                node.name(),
                a, b, 
                get_attribute(node, "input0_dynamic_range").f(),
                get_attribute(node, "input1_dynamic_range").f(),
                node.output(0), 
                get_attribute(node, "precision").s() == "int8" ? Precision::Int8 : Precision::Float16,
                get_attribute(node, "output_precision").s() == "int8" ? Precision::Int8 : Precision::Float16
            );
            tensor_map_by_name[node.output(0)] = n->output(0);
            
       I print  input0_dynamic_range  =0,  input1_dynamic_range  =0   are they right?   tensorRT version 8.4.1.6

NotOnlyAI, 2024-09-27:
find it !     spconv::set_verbose(enable_timer_)  will  give the wrong result !!!

hopef, 2024-09-29:
Could you provide me with a minimal reproduction code?