NVIDIA-AI-IOT/Lidar_AI_Solution #295: CUDA Runtime error cudaMemsetAsync(output_feature_, 0x00, volumn_output_ * sizeof(half), _stream) # an illegal memory access was encountered, code = cudaErrorIllegalAddress [ 700 ] in file CUDA-BEVFusion/src/bevfusion/camera-bevpool.cu:116

Icecream-blue-sky, 2024-11-18:
I try to deploy my own bevfusion model with your code. I only change some hyperparameters like  num cameras and point_cloud_range.  And it raise the error when I run "bash tool/run.sh". 
[image]

And I directly use float32 onnx (skip ptq quantation). There isn't any error in onnx export and tensorrt. Once more, I train my model based on mmdetection3d (https://github.com/open-mmlab/mmdetection3d/tree/main/projects/BEVFusion）. My model config is in the attach file. The coordinate is RFU (Right x, Front y, Up z, which is the lidar coordinate).
[bevfusion_lidar-cam_voxel0075_second_secfpn_8xb4-cyclic-20e.txt](https://github.com/user-attachments/files/17798731/bevfusion_lidar-cam_voxel0075_second_secfpn_8xb4-cyclic-20e.txt)

The log after running "bash tool/run.sh" is in debug.log 
[debug.log](https://github.com/user-attachments/files/17798750/debug.log)

I'd be more than happy to provide you with any other info you want to know. I have been stucked by this for several days. Please help. Thanks.

Icecream-blue-sky, 2024-11-19:
I find bev_height = param.geometry.geometry_dim.x, bev_width = param.geometry.geometry_dim.y. But in mmdetection3d, it's opposite. (https://github.com/open-mmlab/mmdetection3d/blob/fe25f7a51d36e3702f961e198894580d83c4387b/projects/BEVFusion/bevfusion/depth_lss.py#L143) I don't know if this will cause any bug. But after exchanging the bev_height and bev_width, it doesn't change anything.

Icecream-blue-sky, 2024-11-19:
I  found the model I trained is FP32. The BEVFusion in  this project only support FP16. After transforming to FP16, I can infer successfully. But I still want to know if the order of the bev_height and bev_width will influence the performance.

> I find bev_height = param.geometry.geometry_dim.x, bev_width = param.geometry.geometry_dim.y. But in mmdetection3d, it's opposite. [...]

closed 2024-12-11 by Icecream-blue-sky

phoenares, 2025-01-10:
@Icecream-blue-sky hi, 
how did you convert the BEVFusion model trained with mmdetection3d into ONNX and TensorRT? Did you refer to [this link](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/README.md)?  thanks.