NVIDIA-AI-IOT/Lidar_AI_Solution #302: When kernel_size = (1,1,1) for SparseConv3d, the onnx inference result is different from the torch

Icecream-blue-sky, 2024-12-18:
I change the last sparse conv of bevfusion from the first picture to the second (I don't want to downsample):
[image]

[image]

It will cause the final lidar feat to be incorrect. The onnx inference result is inaccurate while the torch inference is normal. I think is probably a bug...

hopef, 2024-12-23:
Please clean the argument "indice_key" because the rulebook needs to be recomputed.

Icecream-blue-sky, 2024-12-24:
> Please clean the argument "indice_key" because the rulebook needs to be recomputed.

This has nothing to do with indice_key. I tried or set indice_key="", indice_key="spconv_4", indice_key="spconv_down2", all do not work. However, after using 1*1*3 conv, I succeeded.

hopef, 2024-12-25:
Could you share a minimal reproduced code with me?

Icecream-blue-sky, 2024-12-25:
> Could you share a minimal reproduced code with me?

Unfortunately, I cannot provide it, because I use BEVFusion from mmdetection3d and have customized it, and I have solved the problem. I just hope that others can try my solution when they encounter similar problems.

closed 2024-12-25 by Icecream-blue-sky