NVIDIA-AI-IOT/Lidar_AI_Solution #303: bevfusion cuda12 libspconv support？

xiongda777, 2024-12-19:
I use orin jeston6.0 where cuda=12.2. can you provide libspconv aarch64 cuda12？  thank you

Tony-syr, 2025-02-25:
Me too, I use orin jeston6.0 where cuda=12.2., could you please provide libspconv aarch64 cuda12? should I create it by myself? and how?

xiongda777, 2025-02-25:
> Me too, I use orin jeston6.0 where cuda=12.2., could you please provide libspconv aarch64 cuda12? should I create it by myself? and how?

hello,I found this project tag deepsteam support cuda12 arm libspconv.so,but its speed is very slow.I test in cuda11 about 55ms ,but this libspconv.so about 110ms.  also you can refer https://github.com/riteshkhrn/Lidar_AI_Solution/tree/main/libraries/New3DSparseConvolution
to build your jeston6.0 lispocnv.so, BUT I failed .If you build success ,please share your libspconv.so cuda12 arm,this is very useful.

Tony-syr, 2025-03-03:
> hello,I found this project tag deepsteam support cuda12 arm libspconv.so,but its speed is very slow.I test in cuda11 about 55ms ,but this libspconv.so about 110ms. also you can refer [...]

I'm trying to create libspconv.so for cuda12 arm, It may needs some time. I will provide it's ready.
Could you please tell me how to use the one in Deepstream? I copy it from the Deepstram branch bu it didn't work. Does it need to copy anything else with it.

123former, 2025-05-14:
> I'm trying to create libspconv.so for cuda12 arm, It may needs some time. I will provide it's ready. Could you please tell me how to use the one in Deepstream? I copy it from the Deepstram branch bu [...]

hello，have you successfully built libspconv for aarch64 with CUDA 12? Could you share it with me, please?

xiongda777, 2025-05-28:
> hello，have you successfully built libspconv for aarch64 with CUDA 12? Could you share it with me, please?

refer to this repo:  https://github.com/xiongda777/bevfusion_ros.git