NVIDIA-AI-IOT/Lidar_AI_Solution #305: is libspconv.so support spconv.SparseInverseConv3d?

wugoukanle, 2024-12-24:
when i want to speed up minkunet，i will meet spconv.SparseInverseConv3d op，is libspconv.so support spconv.SparseInverseConv3d?

cama36, 2025-01-09:
Similar question, in my usage, I will use spconv.SparseConvTranspose3d, I also want to confirm whether this operation is supported? Anyone's answer will be of great help to me, thank you very much for the answer

lisusheen, 2025-02-13:
Is there any update for spconv.SparseInverseConv in spconv.so. When using SparseInverseConv for inferencing, error in the later merge layer [ Different input numel(64 != 128) is invalid when used for add operation]

lisusheen, 2025-02-13:
> Is there any update for spconv.SparseInverseConv in spconv.so. When using SparseInverseConv for inferencing, error in the later merge layer [ Different input numel(64 != 128) is invalid when used for add operation]

Assert failed 💀. a.features.numel == b.features.numel in file src/spconv/engine.cu:596, message: