NVIDIA-AI-IOT/Lidar_AI_Solution #306: To build CUDA-BEVFusion, which version of libprotoc should I install?

Errorsteved, 2025-01-09:
I am trying to build the CUDA-BEVFusion. And installed libprotoc 3.6.1 as stated in this link.https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/19#issuecomment-1556645955
```
(lidarenv) errorsteved@Steve:~/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion$ protoc --version
libprotoc 3.6.1
```
But when I run ```bash tool/run.sh```
It raised terminal error:
```Try to get the current device SM
Current CUDA SM: 86
Configuration done!
-- spconv_lib = ../libraries/3DSparseConvolution/libspconv/lib/x86_64_cuda11.4
-- Configuring done
-- Generating done
-- Build files have been written to: /home/errorsteved/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion/build
[  3%] Linking CXX shared library libcustom_layernorm.so
[ 11%] Building CXX object CMakeFiles/bevfusion_core.dir/src/onnx/onnx-ml.pb.cpp.o
[ 11%] Building CXX object CMakeFiles/bevfusion_core.dir/src/bevfusion/lidar-scn-onnx-parser.cpp.o
[ 14%] Building CXX object CMakeFiles/bevfusion_core.dir/src/onnx/onnx-operators-ml.pb.cpp.o
[ 18%] Built target custom_layernorm
In file included from /home/errorsteved/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion/src/onnx/onnx-operators-ml.pb.cpp:4:
/home/errorsteved/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion/src/onnx/onnx-operators-ml.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/errorsteved/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion/src/onnx/onnx-ml.pb.cpp:4:
/home/errorsteved/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion/src/onnx/onnx-ml.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:3488: CMakeFiles/bevfusion_core.dir/src/onnx/onnx-ml.pb.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:3501: CMakeFiles/bevfusion_core.dir/src/onnx/onnx-operators-ml.pb.cpp.o] Error 1
In file included from /home/errorsteved/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion/src/bevfusion/lidar-scn-onnx-parser.cpp:25:
/home/errorsteved/dev/lidar_ai_solution/Lidar_AI_Solution/CUDA-BEVFusion/src/onnx/onnx-ml.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:3449: CMakeFiles/bevfusion_core.dir/src/bevfusion/lidar-scn-onnx-parser.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:134: CMakeFiles/bevfusion_core.dir/all] Error 2
make: *** [Makefile:84: all] Error 2```

The ChatGPT said that "port_def.inc" only in the last 3.12.0 version.
So which version of libprotoc should I install?

Errorsteved, 2025-01-09:
I found that issuecomment was in 2023 and the repository had been updated since then. Protoc >3.12.0 is sufficient.

closed 2025-01-09 by Errorsteved

zhuo-limeade, 2025-07-10:
Now it needs Protoc >=3.21.0.