NVIDIA-AI-IOT/Lidar_AI_Solution #309: reformat.cpp::executeCutensor::388 checkMacros.cpp::catchCudaError::202

leslie27ch, 2025-01-15:
Hello, I'm really at a loss here and would greatly appreciate your guidance:
I trained an epoch 2 model (using CUDA 11.3), performed PTQ quantization, and converted it to ONNX. When I built and ran it (using CUDA 11.6), everything was normal, and the result visualization was also fine.
Then, I used the epoch 6 model, and without changing anything else, I followed the same steps, but when I ran it (using CUDA 11.6), I encountered the following error. it lools error happened during transfusionhead forward。
[image]

pacientes, 2025-02-07:
@leslie27ch have you solve it?

leslie27ch, 2025-02-14:
> [@leslie27ch](https://github.com/leslie27ch) have you solve it?

I resolved the issue because when computing geometry in BEVPool, I made a mistake in modifying the "kept" condition. I originally intended not to calculate the area within 5 meters in front of and behind the ego vehicle (自车前后各5米范围), but the implementation logic was incorrect.