NVIDIA-AI-IOT/Lidar_AI_Solution #319: Datatype shows "Assert failed 💀" after reconfig layer precision

pengyusi, 2025-05-13:
使用CUDA Centerpoint进行推理时，因为量化后的scn模型出现比较严重的问题，怀疑是精度损失问题，因此尝试逐层用fp16精度debug。当我设置第二个block为fp16时，推理报错提示；数据类型不符。以下是报错内容：
```Assert failed 💀. a.features.dtype() == b.features.dtype() in file src/spconv/engine.cu:600, message: [add1] Different input dtype(Float16 != Int8) is invalid when used for add operation```
以下是我重新配置的fp16精度层：
```
def disable_quant_layer(model_finetune): 
    print("🔥Disable Conv1FirstSparseBasicBlock Mode🔥") 
    quantize.disable_quantization(model_finetune.conv_input).apply()
    quantize.disable_quantization(model_finetune.encoder_layers[0][0].conv1).apply()
    quantize.disable_quantization(model_finetune.encoder_layers[0][0].conv2).apply()
    quantize.disable_quantization(model_finetune.encoder_layers[0][0].quant_add).apply()
    quantize.disable_quantization(model_finetune.encoder_layers[0][1].conv1).apply()
    quantize.disable_quantization(model_finetune.encoder_layers[0][1].conv2).apply()
    quantize.disable_quantization(model_finetune.encoder_layers[0][1].quant_add).apply()
    return model_finetune

if __name__ == "__main__":
        #some code
        for name, module in model.named_modules():
            module.precision = "int8"
            module.output_precision = "int8"
        model.conv_input.precision = "fp16"
        model.encoder_layers[0][0].conv1.precision = "fp16"   
        model.encoder_layers[0][0].conv2.precision = "fp16"
        model.encoder_layers[0][0].quant_add.precision = "fp16"
        model.encoder_layers[0][1].conv1.precision = "fp16"   
        model.encoder_layers[0][1].conv2.precision = "fp16"
        model.encoder_layers[0][1].quant_add.precision = "fp16"
        model.conv_input.output_precision = "fp16" 
        model.encoder_layers[0][0].conv1.output_precision = "fp16"   
        model.encoder_layers[0][0].conv2.output_precision = "fp16"
        model.encoder_layers[0][0].quant_add.output_precision = "fp16"    
        model.encoder_layers[0][1].conv1.output_precision = "fp16"   
        model.encoder_layers[0][1].conv2.output_precision = "fp16"
        model.encoder_layers[0][1].quant_add.output_precision = "int8"    
```
请问可能的原因是什么？

hopef, 2025-05-18:
因为要确保add算子的左右两侧是相同数据类型的。确认一下输入给add的两个input是不是都是int8或者fp16.

pengyusi, 2025-05-20:
> 因为要确保add算子的左右两侧是相同数据类型的。确认一下输入给add的两个input是不是都是int8或者fp16.

确认过，逻辑上应该是一致的，onnx如下：

[image]

应该是红框的算子报错，前序层全部设置不使用量化，但是模型推理是配置了int8加载的。会不会是解析onnx的库对Relu的处理只判断了模型加载有没有用int8,导致add过后的数据格式变了？

另外我在替换另一个模型时报错：`Assert failed 💀. false in file src/spconv/engine.cu:840, message: No tactic was used to inference` 请问这个具体是什么意思呢，我对照onnx的主要参数，和其他能跑的模型没有明显区别啊 🙇‍♂