NVIDIA-AI-IOT/Lidar_AI_Solution #321: no kernel image is available for execution on the device （libspconv.so (aarch64_cuda12.8)）

123former, 2025-05-14:
Thank you for releasing libspconv.so (aarch64_cuda12.8). An error occurred when executing "bash tool/run.sh".
[image]

[rulebook.cu:273] (Error) CUDA Runtime error spconv::setup_hash_and_route<<<(num_input + 1023) / 1024, 1024, 0, stream>>>( hash_.get(), route_mask_.ptr(), route_.ptr(), indices, num_input, prob->kv) # no kernel image is available for execution on the device, code = cudaErrorNoKernelImageForDevice [ 209 ] in file src/spconv/rulebook.cu:273
tool/run.sh: line 44:  5816 Aborted                 (core dumped) ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION

123former, 2025-05-14:
[image]

hopef, 2025-05-18:
I have uploaded the libspconv that supports SM87. Please give the new one a try. Thank you!

zjh1378805302, 2025-12-12:
> You can find a suitable libspconv.so for sm_87 in another branch, not the master one. It works.

Okay, thank you. It has run successfully. If I want to test its image detection performance now, do you have any suggestions?

cheriku, 2025-12-12:
> > > > You can find a suitable libspconv.so for sm_87 in another branch, not the master one. It works.
> > > 
> > > 
> > > Okay, thank you. It has run successfully. If I want to test its image detection performance now, do you have any suggestions?
> > 
> > 
> > I’m also working with Orin and get some problems. May I ask which libspconv.so you are using?
> 
> What is your CUDA version? I am using the previous version using CUDA12.2, and I have not yet found a suitable. so file on CUDA12.6

I am using CUDA12.6